globsky/SignalSim #10: What features are needed for IF generation ? Can we help somehow ?

bangpradyumna, 2025-02-13:
Hello,
I understand that IF generation is scheduled for Q1 2025 but, I was wondering what all things are required to be added in current code for IF generation capability ?

If you could mention those points, maybe we can contribute towards those goals as well. Also, wondering if you've already started the work on it ?

Thanks again for this amazing project!

globsky, 2025-02-15:
Thank you for interested in my project.
To achieve IF signal generation, the following procedure is required:
1. Update satellite parameters at a certain interval (eg. 1ms), this is shown in the behavior model implementation of my GNSS baseband project, you can find it in https://github.com/globsky/greta-oto/blob/main/SimModel/src/GnssTop.cpp function int CGnssTop::StepToNextTime() 
2. Use CSatelliteSignal class implemented in SatelliteSignal.cpp to get data modulated data and pilot channel signal (including nominal phase and amplitude) at current epoch. The example usage of this class can be found in https://github.com/globsky/greta-oto/blob/main/SimModel/src/TrackingChannel.cpp
3. Apply PRN code to each interpolated IF sample (PRN code generation has been implemented but not committed yet)
4. Interpolate carrier phase of each IF sample using start and end carrier within this period
5. Multiply each satellite IF signal with corresponding amplitude and add them together, then add noise
6. Quantize sample and write to file
I know this involves a lot of work. I am still working on it and a basic framework has been implemented.

sliencefade, 2025-03-05:
Hello, I am now simulating the Beidou signal based on the GPS-SDR-SIM project, and now I have realized the signal generation of B1I and B3I, and completed the verification using GNSS-SDR and on-board chips. Right now, I'm working on B2b coding and I'm almost done. You have a lot of research in the field of navigation signal simulation, can you leave a contact information? There are some questions that need to be asked.

globsky, 2025-03-10:
Any question you can send email to globskytech@gmail.com

closed 2025-04-30 by globsky