globsky/SignalSim #21: I managed to launch GPS L1CA + Galileo E1 + BeiDou B1I

danusha2345, 2025-06-09:
[image]

all three systems are determined by the receiver. I'm broadcasting via hackrf

[image]

xingxingzaoqi, 2025-06-10:
May I ask if you have made any modifications to the code? I am unable to enable B1I.

danusha2345, 2025-06-10:
> May I ask if you have made any modifications to the code? I am unable to enable B1I.

https://dropmefiles.com/vGJKh

meressekyalwe, 2025-06-12:
Круто! спасибо большое @danusha2345

meressekyalwe, 2025-06-12:
Я только что запустил код — вроде бы внесены не все изменения. @danusha2345

danusha2345, 2025-06-26:
> [@danusha2345](https://github.com/danusha2345) Please send me the modifications of the code, the link is broken ?? also could you please give me the config you have used. ^-^

https://dropmefiles.com/hWH41

embeddedteam2025, 2025-06-29:
what is the config file for simulate the three gnss constellations GPS, BDS and GAL ? and if it is possible did you use more than one hackRF to transmit the 3 GNSS Constellations simultaneously.

danusha2345, 2025-06-30:
у меня в репозитории есть ветка https://github.com/danusha2345/SignalSim/tree/ifdatagen-standalone-improvements