globsky/SignalSim #26: Unable to see satellites in any config

bangpradyumna, 2025-06-20:
Hi, I compiled the code on Ubuntu system and when I did 

```
./IFdataGen ../configs/IfGenTest.json
Generate IF data with following satellite signals:
GPS visible SVs 0.
BDS visible SVs 0.
Galileo visible SVs 0.
Glonass visible SVs 0.

GPS L1CA with IF -4580kHz
```

This is same for all the configs present in the ./configs/ folder. Not sure what I am doing wrong since the files are as provided in the main branch of this repo, I've not added or changed anything. Any pointers to help me debug this ?

Thanks in advance.

globsky, 2025-06-20:
Check the path of the RINEX ephemeris file in JSON config file. The directory is relative to the place you execute the command, not relative to the path of the JSON file. You may need to run it at the parent directory according to the command you provided.

bangpradyumna, 2025-06-20:
I'd also like to add the installation instructions for Linux.

globsky, 2025-06-20:
The reason that the CNavData::ReadNavFile() or its caller does not terminate or throw an error when ephemeris file open error is that the configuration allows multiple RINEX files be loaded (for cases that the navigation file is not a merged one, eg. several files from different time or constellation). And one file load error should not terminate the whole process. The functions in itself should not determine the work flow. Instead it should be determined by the user application. It is better for the SignalSim library itself to throw errors or warnings of different type. But I have not found out a good way to do it.

qaisarudemy, 2025-06-21:
Within few days next update is coming, with faster signal generation time, probably 1000x than the current version, but still not real time. Further Linux build steps and stand-alone build without MVS would also be included.

bangpradyumna, 2025-06-23:
@globsky Instead of the ReadNavFile function, what if the checks are added to SetEphermerisFile function as follows : 

`BOOL SetEphemerisFile(JsonObject *Object, CNavData &NavData)
{
    FILE *fp = NULL;
	while (Object)
	{
		if (strcmp(Object->Key, "name") == 0)
        {
            if ((fp = fopen(Object->String, "r")) == NULL)
            {
                std::cerr << "Error: Cannot open Ephemeris file: " << Object->String << std::endl;
                return FALSE;
            }
            fclose(fp);
			NavData.ReadNavFile(Object->String);
        }
		Object = JsonStream::GetNextObject(Object);
	}
	return TRUE;
}`

Wouldn't this solve the issue ?