globsky/SignalSim #30: Issue with L2C Signal Generation

meressekyalwe, 2025-06-27:
L2C is not working. Also, some parameters need to be checked in PrnGenerate.

danusha2345, 2025-06-27:
[image]

danusha2345, 2025-06-27:
там отсутсвует break, но не у gps, а у галилео

danusha2345, 2025-06-27:
у меня пока нет мультидиапазонного приёмника, скоро приедет , будут проверять все сигналы.

globsky, 2025-06-28:
https://www.gps.gov/technical/icwg/IS-GPS-200M.pdf is the ICD including L2C signal. I have a fix in the latest update for L2C but has not tested it yet.

globsky, 2025-07-08:
L2C should be OK from Commit 88de92a. At least correlation result show obvious peak with Doppler and SVID provided in output message when using MATLAB to analysis the IF data file.

closed 2025-08-13 by globsky