globsky/SignalSim #35: I cant build a project

Mysteo91, 2025-06-30:
Hello ! I tried to build your project with GCC 13.3.0 but got error :
```
 undefined reference to `FastMath::cos_lut'
 undefined reference to `FastMath::sin_lut'
 undefined reference to `FastMath::lut_initialized'
```
How i can fix it ?

Mysteo91, 2025-06-30:
О,  хоть кто-то рускоговорящий попался. **[danusha2345](https://github.com/danusha2345)** а этим пакетом можно сэмулировать GLONASS сигнал и посмотреть ловит ля GPS модуль его ?  Для трансляции есть Pluto SDR

Mysteo91, 2025-06-30:
> https://github.com/danusha2345/SignalSim/tree/ifdatagen-standalone-improvements
> 
> Try my branch. there may have been incorrect commits

Ваш форк сбилдился, но после выполнения 
```
./IFdataGen/IFdataGen presets/GLO_G1_only.json
```
Никаких новых файлов или папки generated_files не появилось

danusha2345, 2025-06-30:
передавать надо с сэмплрэйтом который использован в пресете, в глонассе 10мгц, и ширина полосы 10 мгц. у меня нет проблем с 10 мгц. у меня b210

hassannawaz1210, 2025-07-03:
> Hello ! I tried to build your project with GCC 13.3.0 but got error :
> 
> ```
>  undefined reference to `FastMath::cos_lut'
>  undefined reference to `FastMath::sin_lut'
>  undefined reference to `FastMath::lut_initialized'
> ```
> 
> How i can fix it ?

This is because **FastMath.cpp** was missing from CMakeLists.txt. As the CMake code has been improved and made more generic, it shouldnt be a problem now.

closed 2025-08-14 by globsky