globsky/SignalSim #39: No constellation output in Pocket SDR GUI using IQ samples generated from SignalSim (L1CA, latest version)

eebushra, 2025-07-08:
Hi,
I generated GPS L1CA IQ samples using the latest version of SignalSim. I followed the standard procedure and used a JSON configuration file with the following key parameters:

    Signal type: L1CA

    Sample rate: 4.0 MHz

    Center frequency: 1575.42 MH

[IfGenTest.json](https://github.com/user-attachments/files/21114927/IfGenTest.json)

    Output file: "IfGenTest8.bin"

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

After successful generation, I wanted to test only the baseband signal, so I opened the generated .bin file using Pocket SDR GUI. I selected File mode (instead of RTL-SDR or other hardware) and used the following Pocket SDR command-line parameters or GUI settings:

    Format: INT8X2

    IQ Order: IQ

    Sample Rate: 4.0e6

    Frequency: 0e6

    Signal: L1CA

    PRNs: 1–32 (tested various PRNs)

However, when I launch the GUI, I don't see any constellation output or signal lock. The GUI remains blank, as if it's not detecting any valid signal or acquisition fails.

Questions:

    Are there any additional configuration steps required to make the Pocket SDR GUI compatible with the generated samples?

    Could this be due to missing metadata, incorrect format, or signal scaling?

    Has anyone successfully used SignalSim-generated IQ samples with Pocket SDR?

Any guidance or example JSON files that are known to work with Pocket SDR would be appreciated.
Thanks in advance!

danusha2345, 2025-07-08:
[image]
работает.

LITIECX, 2025-07-08:
"output": {
		"type": "IFdata",
		"format": "IQ8",
		"sampleFreq": 4,
		"centerFreq": 1575.42,
		"name": "IfGenTest8.bin",
		"config": {
			"elevationMask": 3
		},
"format": "IQ8",   amend  "format": "IQ4",
Refer to this modification：
https://github.com/osqzss/gps-sdr-sim/blob/master/gpssim.c#L2395

MuhammadQaisarAli, 2025-07-08:
> I HAVE MADE THIS TAG FILE AS WELL FOR POCKET SDR i have also tested by setting bits 8 for IQ8 format

Just changing the tag file data won't make the `PocketSDR to process signal properly, since sample depth and bits formatting of PocketSDR may be different from that of used by `SignalSim`. The samples should be properly written to bin file.

eebushra, 2025-07-18:
issue resolved https://github.com/tomojitakasu/PocketSDR/issues/55

closed 2025-07-18 by eebushra