globsky/SignalSim #40: Testing GLONASS G1 There is only signal but no time. What is the problem?

LITIECX, 2025-07-08:


danusha2345, 2025-07-08:
Глонасс пока что не работает. Точно работают только l1ca, b1i, b1c, e1

MuhammadQaisarAli, 2025-07-08:
> Попробуй мой форк. У меня e1 работает

i just tried https://github.com/danusha2345/SignalSim/tree/if_data_danusha, even it doesn't detect visible sats.

```
SignalSim$ ./IFdataGen/IFdataGen presets/GPS_BDS_GAL_L1_5MHz.json

================================================================================
                          IF SIGNAL GENERATION
================================================================================
[INFO]  Loading JSON file: presets/GPS_BDS_GAL_L1_5MHz.json
[INFO]  JSON file read successfully: presets/GPS_BDS_GAL_L1_5MHz.json
[INFO]  Loading ephemeris file: EphData/rinex_v3_20251560000.rnx
[INFO]  Ephemeris file loaded successfully: EphData/rinex_v3_20251560000.rnx
[INFO]  Opening output file: generated_files/GPS_BDS_GAL_L1_5MHz_3min.C8
[INFO]  Output file opened successfully.
[INFO]  OpenMP configured for PARALLEL execution (20 threads auto-detected)
[INFO]  Generating IF data with following satellite signals:

[INFO]  Enabled Signals:
        GPS : [ L1CA L1C ]
        BDS : [ B1C ]
        GAL : [ E1 ]

Signals Summary Table:
+---------------+-------------+--------------+------------------------------+
| Constellation | Visible SVs | Signals / SV | Total Signals / Visible SVs |
+---------------+-------------+--------------+------------------------------+
| GPS           | 0           | 2            | 0                            |
| BeiDou        | 0           | 1            | 0                            |
| Galileo       | 0           | 1            | 0                            |
| GLONASS       | 0           | 0            | 0                            |
+---------------+-------------+--------------+------------------------------+
Total Visible SVs = 0, Total channels = 0

GPS L1CA with IF +0kHz:
+----+--------------+----+--------------+----+--------------+----+--------------+
| SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) |
+----+--------------+----+--------------+----+--------------+----+--------------+
+----+--------------+----+--------------+----+--------------+----+--------------+

GPS L1C with IF +0kHz:
+----+--------------+----+--------------+----+--------------+----+--------------+
| SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) |
+----+--------------+----+--------------+----+--------------+----+--------------+
+----+--------------+----+--------------+----+--------------+----+--------------+

BeiDou B1C with IF +0kHz:
+----+--------------+----+--------------+----+--------------+----+--------------+
| SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) |
+----+--------------+----+--------------+----+--------------+----+--------------+
+----+--------------+----+--------------+----+--------------+----+--------------+

Galileo E1 with IF +0kHz:
+----+--------------+----+--------------+----+--------------+----+--------------+
| SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) | SV | Doppler (Hz) |
+----+--------------+----+--------------+----+--------------+----+--------------+
+----+--------------+----+--------------+----+--------------+----+--------------+

[INFO]  Starting signal generation loop...
[INFO]  Signal Duration: 180.00 s
[INFO]  Signal Size: 1716.61 MB
[INFO]  Signal Data format: IQ8
[INFO]  Signal Center freq: 1575.4200 MHz
[INFO]  Signal Sample rate: 5.00 MHz

[=>                     2.3%                       ] 4220/180000 ms | 40.25/1716.61 MB | 12.37 MB/s | ETA: 2m15s   ^C

``

LITIECX, 2025-07-09:
> Глонасс пока что не работает. Точно работают только l1ca, b1i, b1c, e1

I see you have made a lot of changes to G1 (https://github.com/danusha2345/SignalSim/tree/if_data_danusha). Do you have any plans to solve the problem of G1 not having enough time?

closed 2025-11-20 by globsky