globsky/SignalSim #43: GLONASS L3

mehdi200180, 2025-07-11:
Hi. why did you not glonass L3 signal? could you find it's ICD in the internet (because i have not found it and i thought maybe that is the reason you have not implemented it).
another question, is it possible to simulate Iridium signal?
and the last question, how can i get a commercial license for seeing L2P and L1P generation source code?

globsky, 2025-07-11:
An official contract need to be signed for commercial use of the code and any other add-on services. Globsky Technology Inc. is an entity registered in California USA to provide commercial license.
PS: Documents similar to ICD for Iridium STL signal is welcome if you have link to download it.

mehdi200180, 2025-07-12:
Thanks.
what about Glonass L3. why have you not implemented it?

globsky, 2025-07-12:
GLONASS L3 is not in recent plan.